ArminJo/Talkie #1: how to add new words ?

marwan211, 2019-09-05:
can you make a video tutorial to make talkie library speak specified words ?

HectorOrdonez, 2020-04-07:
@marwan211 Did you figure this out eventually?

ArminJo, 2020-04-07:
Not yet

ahmedebeed555, 2020-08-11:
Finally I could generate voices. Thanks all for your help.
[https://youtu.be/KQseCA0nftI](https://youtu.be/KQseCA0nftI
)

christurnerchris, 2020-11-15:
Yes it worked windows  incredible 
Just need a Linux version

ArminJo, 2020-11-15:
@ahmedebeed555 I had no time to test it by myself, but I now added  a link to your video in the readme, many thanks 🥇

christurnerchris, 2020-11-15:
tried it in  windows 10 . 
downloaded some samples from net - converted to wav using goldwave, 16 bit mono signed and 8bit mono both produce output but noise is mainly gibberish.  single piano notees just make terrible sound
Recorded my own voice (computer microphone) and quality is good.  think its background noise confuses process
amazing that anything works really !

ahmedebeed555, 2020-11-15:
> @ahmedebeed555 I had no time to test it by myself, but I now added a link to your video in the readme, many thanks 🥇

Thank you very much.

christurnerchris, 2020-11-15:
Got python_wizard_gui.py  to run in pyroid3 on Android
I don't know how to put the -f arduino on the end but it has defaulted to Arduino anyway!