ArminJo/Talkie #4: no output using teensy 3.2

gretel, 2020-05-23:
version 1.0.2 (1.0.3 not released?) gives no output. tried changing pin declaration manually without effect.
the original library from paul stoffregen works fine, on pin `A14` (`DAC`).
i'd be glad if this library could work as a drop-in-replacement. thank you!

ArminJo, 2020-05-23:
I will have a look.

ArminJo, 2020-05-28:
I installed with tensiduino installer and there is no original talkie library 😢 
But my sources are compiling and I made a few changes. But I am not finished yet.

gretel, 2020-05-28:
uh "original" was a bit far fetched.. this one

https://github.com/PaulStoffregen/Talkie

ArminJo, 2020-06-05:
I uploaded the "final" version for Teensy.
Can you please test it and give feedback.
Then I can build a new Arduino library release.

ArminJo, 2020-06-06:
I just fixed a bug and updated the repo.

closed 2020-06-14 by ArminJo

gretel, 2020-06-22:
@ArminJo thank you. going to report back.