ArminJo/Talkie #7: How do I make the program dynamic

suraj-b, 2020-07-15:
I have been trying to use this library and make it dynamic. What I'm trying to do is receive data from another arduino, say the keyword *DANGER*, and pass it through the in-built function voice.say(). I do understand that the voice.say() accepts uint_8t datatype. Any help is greatly appreciated.

ArminJo, 2020-07-15:
Best is you study the code, then you will find the answers.

suraj-b, 2020-07-16:
Is there a way I could pass the arguements to voice.say() using a loop?

suraj-b, 2020-07-16:
I think what we can do is take the address of the array, lets say of DANGER,  from program memory and map the address to its respective word. Whenever a word is detected in user input we can point to the address in voice.say(). Can this be done?

ArminJo, 2020-07-16:
Of course this can be done, but keep in mind that every word is precompiled and can not created at runtime. Takie is NO text to speech engine.

closed 2020-07-16 by ArminJo