ArminJo/Talkie #10: Use regular PWM on Teensy

grapefrukt, 2020-08-25:
Hey, 

I'm trying to make this library play nice with the Teensy Audio library, which I have set up to use the DAC on my Teensy 3.2. 

It's not especially clear to me how I initialise Talkie to stay away from that pin and use a regular old PWM pin. I can use `beginPWM()`, but it also seems that `initializeHardware()`which runs before anything it output (I think?) is [eager to use the DAC anyway](https://github.com/ArminJo/Talkie/blob/3071ef4f55e7d9cfe2eef1df3f776c70d1b2e0db/src/Talkie.cpp#L299).

Is there a way to do this?

(Ideally, I'd mix the Talkie output into the Teensy Audio library, but that's a bridge I'll cross another day)

ArminJo, 2020-08-25:
Try to outcomment line 125 ff.
```
#  if defined(__MKL26Z64__) // Teensy LC
#  define DAC_PIN A12
```
and look at line 259 ff.
`#elif defined(TEENSYDUINO)`

grapefrukt, 2020-08-28:
I decided to sidestep this issue and use samples instead, so I won't be digging deeper into this. 
Thanks for your help so far and sorry for bothering you!

closed 2020-08-28 by grapefrukt