ArminJo/Talkie #11: talkie and Tone wont load in same sketch

christurnerchris, 2020-11-14:
Armin .Fantastic work - &  you read my mind .  
I want to use Talkie and tone or Pitch to have  words and music on a project (Im re using an old super simon game and I want it to make music and talk )  . 
I am using chinese clone UNO and  NANO   
I tried this : 
#include <Arduino.h>
//   #include <TalkieLPC.h>
#include <Vocab_US_Acorn.h>
#include <Vocab_US_Large.h>
#include <Vocab_US_TI99.h>
#include <Talkie.h>
#include <Vocab_Special.h>
#include <digitalWriteFast.h>
#include <Vocab_US_Clock.h>
#include <TalkieUtils.h>
#include <Vocab_Toms_Diner.h>
#include <Vocab_Soundbites.h>

    #include "Tone.h"

I got this error
 Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `timer0_pin_port'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `timer0_pin_mask'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `timer1_pin_port'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `timer1_pin_mask'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `timer2_pin_port'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `timer2_pin_mask'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `timer0_toggle_count'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `timer1_toggle_count'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `timer2_toggle_count'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `__vector_7'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Uno.

christurnerchris, 2020-11-14:
heres my bright idea  
How about making a new vocab for Talkie with the notes as a  buzzer , pure sine wave and piano  say for all the notes 
then you wont have make it compatible with tthe Tone.h library  
Ive watched  the video by the other commenter looks complicated  - can you use the python in  Raspberry pi  ?

ArminJo, 2020-11-14:
Why do you include Tone.h explicitely??? This may be the error.

christurnerchris, 2020-11-15:
Ok I'll try that .
I have got the python method to work (upto getting programmed page to load) using instructions at 
https://youtu.be/KQseCA0nftI
Is that your channel?

ArminJo, 2020-11-15:
I can use tone() and Talkie without any problems but not simultaneous, since they use the same timer.
see the [PlayChristmasMelodyUSDistance example](https://github.com/ArminJo/PlayRtttl/blob/master/examples/PlayChristmasMelodyUSDistance/PlayChristmasMelodyUSDistance.ino).

> Is that your channel?

Its not my channel, but I now included a link to it.

closed 2020-11-26 by ArminJo