ArminJo/Talkie #13: Distorted and drops first consonant

gewoods3, 2021-04-18:
Talkie is distorted and drops first consonant using Adafruit Feather M0.
Sounds like: DANGER DANGER _ED ALERT _OTOR IS SON FIRE

I wonder if the code gets out of the 'synthPeriod' state too early, but I don't see how to prove it?

```c
void timerInterrupt()
...
    if (synthPeriod) {
        ...
    } else {
        // Unvoiced source. Use random bit generator / white noise
        ...
    }
```

* Talkie version 1.2.0 (standard Arduino library)
* Adafruit Feather M0 Basic Proto - ATSAMD21 Cortex M0
* https://www.adafruit.com/product/2772

If you will look at it, I will buy and send you an Adafruit Feather M0.

Thank you for your work.

ArminJo, 2021-04-18:
I can look at it and find it generous, that you will buy me a board 😀 .
But Isn't it the same chip as on the Arduino Zero?
This I have at hand and can test next week.

gewoods3, 2021-04-18:
Thank you for your prompt response. I believe it is the same chip as the Zero, but.. .

gewoods3, 2021-04-18:
https://www.adafruit.com/product/2772

"At the Feather M0's heart is an ATSAMD21G18 ARM Cortex M0 processor, clocked at 48 MHz and at 3.3V logic, the same one used in the new Arduino Zero."

ArminJo, 2021-04-20:
I have no problems with my Zero board. 
https://www.dropbox.com/sh/flu6uexhmdr212e/AADoDr8TG1hjJ9_CxR9mWPEsa?dl=0

gewoods3, 2021-04-20:
Current Master works perfect on Adafruit M0 SAMD21!

closed 2021-04-20 by gewoods3