ArminJo/Talkie #28: [Bug]: demo toms diner: harsh sound

msalat, 2022-05-07:
### The type of board you are using.

Arduino ATmega328* board (UNO, Nano, etc.)

### What IDE are you using?

Arduino IDE

### Example(s) you have checked while hunting the bug.

- [X] Demo_Toms_Diner
- [X] Getting_Started
- [ ] TalkieVocabularyDemo
- [ ] USDistanceToVoice
- [ ] Voltmeter
- [ ] VoltmeterSayQ
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [X] Demo_Toms_Diner
- [ ] Getting_Started
- [ ] TalkieVocabularyDemo
- [ ] USDistanceToVoice
- [ ] Voltmeter
- [ ] VoltmeterSayQ
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [ ] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

Playing this sample with this circuit 
[image]

### What happens?

95 % of the time it sounds clear, but occasionally — and always at the same time — it sounds absolutely harsh.

### What were you expecting to happen?

clear sound as in the *getting started* demo

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

msalat, 2022-05-07:
PS: Speaker is a *visaton fr 10 hm 4 Ω*

msalat, 2022-05-07:
4700 µF capacitor between 5 V and GND does not help.

msalat, 2022-05-07:
Perhaps got such sound on my scoppy osci:
[image]

msalat, 2022-05-07:
I've tried the toneMeldoy example with the same audio output stage and it is much louder and clear.

ArminJo, 2022-05-10:
At which second do you get the harsh sound?

> I've tried the toneMeldoy example with the same audio output stage and it is much louder and clear.

Yes, the toneMeldoy example is a awesome piece of software to get this sound.

msalat, 2022-05-12:
Harsh from 1.04s to 1.2s and every few seconds...
(recorded from my smartphone)
[image]

V205Github, 2023-07-08:
I agree that the sound is bad. I think the problem is with the recording.

richard-broadhurst, 2026-04-26:
The recording is fine on the original talkie project, but the harsh noise comes from the encoding.
If not all the clamping and wrapping is applied, it sounds fine, but if it is you get the harsh bits.
e.g. The old mame code plays fine, the new, more accurate, mame code plays the harsh bits.
I'm still checking if this is to do with using the analogue output vs the digital output but it doesn't look lie it.
Playing it on my MMC Micro also plays the harsh bits, although not identical to the latest mame code but close.