ArminJo/Talkie #40: DEBUG and ESP32 Don't appear to be working

madmacks59, 2023-12-26:
I've installed the library, added a #Define ESP32 and DEBUG, compiled and loaded the code to my HelTec ESP32. I get no display at the console for the info I'd expect (based on the ccp file) and I get no should at all out of pin 25 (or 26 for that matter if I change the ccp file Pin). Is there some magical incantation I need? Or am I just missing something?

ArminJo, 2023-12-26:
Please check the Voltmeter example.

madmacks59, 2023-12-26:
Checking now...

madmacks59, 2023-12-26:
So, looking at the code, I don't see anything that helps answer the question. Running the code simply displays an integer at the console, no sound, nothing else. Is there something else that you are thinking needs checking? I'm running the OOTB code so I'll try with the define ESP32 statement...

ArminJo, 2023-12-26:
An integer output as result of the statement    `Serial.println(F("START " __FILE__ " from " __DATE__ "\r\nUsing library version " VERSION_TALKIE));` is very surprising!

closed 2023-12-26 by ArminJo

madmacks59, 2023-12-26:
To be clear... Here's what I get...

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
START /private/var/folders/dl/rw15y_1n1ds_8b6txcg9_8xw0000gp/T/.arduinoIDE-unsaved20231126-26407-1ib5ju1.nd8ag/Voltmeter/Voltmeter.ino from Dec 26 2023
Using library version 1.3.2
20014
20014
20014
20014

madmacks59, 2023-12-26:
So, I guess you closed the issue... But, I've also done a few things checking hardware and it's starting to look like it might be the amp I'm using (PAM8403). I've hook things up using a standard Arduino Uno and I still don't get any sound. That makes me think it's the Degraw PAM8403 itself. I'll order a replacement and see if that resolves the issue.

ArminJo, 2023-12-26:
You know the Discussions and Q&A section of GitHup repos?