ArminJo/Talkie #46: [Bug]: Unable to get Hiletgo ESP32-Wroom-32D (38 Pin Version) working with Talkie.

madmacks59, 2024-09-04:
### The type of board you are using.

ESP32 board

### What IDE are you using?

Arduino IDE

### Example(s) you have checked while hunting the bug.

- [ ] Demo_Toms_Diner
- [ ] Getting_Started
- [ ] TalkieVocabularyDemo
- [ ] USDistanceToVoice
- [ ] Voltmeter
- [ ] VoltmeterSayQ
- [X] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] Demo_Toms_Diner
- [ ] Getting_Started
- [ ] TalkieVocabularyDemo
- [ ] USDistanceToVoice
- [ ] Voltmeter
- [ ] VoltmeterSayQ
- [X] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

Sketch is using IDE 2.3.2, Arduino ESP32 Boards 2.0.13, ESP32 (Espressif) 3.0.4. 

Pin mapping is:
Right Speaker 27
Left Speaker 26
LED-Red 18
LED-Yellow 5
LED-Green 19
Float Sensor-1 17
Float Sensor-2 16
Float Sensor-3 4

Code measures the level of a 55 gallon barrel using the three float sensors. When various water levels are reached various LEDs are turned on to give visual feedback. In addition, as each float sensor triggers am audio message is played indicating the level of the water in the barrel (tank full, tank 66%, etc).

I don't see any demo code for the ESP32 so I couldn't test it. However, I did have Talkie working on this set up when I was using a TOF sensor instead of the float sensors.

### What happens?

The code fails to compile with the ESP 3.0.4 core library. It throws the compile error show in the next section.

When I change to ESP32 core 2.0.17 it compiles cleanly, but throws the following error at run time.

E (7336) timer_group: timer_isr_callback_add(236): register interrupt service failed

### The serial output which indicates the error happened.

```shell
The code fails to compile with the ESP 3.0.4 core library. Throws:

In file included from /Users/madmacks/Documents/Arduino/libraries/Talkie/src/Talkie.cpp:126:
/Users/madmacks/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-b6b4727c58/esp32/include/driver/deprecated/driver/dac.h:16:2: warning: #warning "The legacy DAC driver is deprecated, please use `driver/dac_oneshot.h`, `driver/dac_cosine.h` or `driver/dac_continuous.h` instead" [-Wcpp]
   16 | #warning "The legacy DAC driver is deprecated, please use `driver/dac_oneshot.h`, `driver/dac_cosine.h` or `driver/dac_continuous.h` instead"
      |  ^~~~~~~
/Users/madmacks/Documents/Arduino/libraries/Talkie/src/Talkie.cpp: In member function 'void Talkie::initializeHardware()':
/Users/madmacks/Documents/Arduino/libraries/Talkie/src/Talkie.cpp:321:40: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
  321 |     sTalkieSampleRateTimer = timerBegin(1, APB_FREQUENCY_DIVIDER, true);
      |                              ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/madmacks/Library/Arduino15/packages/esp32/hardware/esp32/3.0.4/cores/esp32/esp32-hal.h:84,
                 from /Users/madmacks/Library/Arduino15/packages/esp32/hardware/esp32/3.0.4/cores/esp32/Arduino.h:36,
                 from /Users/madmacks/Documents/Arduino/libraries/Talkie/src/Talkie.cpp:78:
/Users/madmacks/Library/Arduino15/packages/esp32/hardware/esp32/3.0.4/cores/esp32/esp32-hal-timer.h:35:13: note: declared here
   35 | hw_timer_t *timerBegin(uint32_t frequency);
      |             ^~~~~~~~~~
/Users/madmacks/Documents/Arduino/libraries/Talkie/src/Talkie.cpp:322:25: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
  322 |     timerAttachInterrupt(sTalkieSampleRateTimer, timerInterrupt, true);
      |     ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/madmacks/Library/Arduino15/packages/esp32/hardware/esp32/3.0.4/cores/esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
/Users/madmacks/Documents/Arduino/libraries/Talkie/src/Talkie.cpp:323:5: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
  323 |     timerAlarmWrite(sTalkieSampleRateTimer, (getApbFrequency() / APB_FREQUENCY_DIVIDER) / SAMPLE_RATE, true);
      |     ^~~~~~~~~~~~~~~
      |     timerWrite
/Users/madmacks/Documents/Arduino/libraries/Talkie/src/Talkie.cpp:324:5: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
  324 |     timerAlarmEnable(sTalkieSampleRateTimer);
      |     ^~~~~~~~~~~~~~~~
      |     timerAlarm
/Users/madmacks/Documents/Arduino/libraries/Talkie/src/Talkie.cpp: In member function 'void Talkie::terminateHardware()':
/Users/madmacks/Documents/Arduino/libraries/Talkie/src/Talkie.cpp:439:5: error: 'timerAlarmDisable' was not declared in this scope
  439 |     timerAlarmDisable(sTalkieSampleRateTimer);
      |     ^~~~~~~~~~~~~~~~~
/Users/madmacks/Documents/Arduino/libraries/Talkie/src/Talkie.cpp: In member function 'uint8_t Talkie::getBits(uint8_t)':
/Users/madmacks/Documents/Arduino/libraries/Talkie/src/Talkie.cpp:537:9: warning: '++' expression of 'volatile'-qualified type is deprecated [-Wvolatile]
  537 |         WordDataPointer++;
      |         ^~~~~~~~~~~~~~~
/Users/madmacks/Documents/Arduino/libraries/Talkie/src/Talkie.cpp: In member function 'void Talkie::FIFOPushBack(const uint8_t*)':
/Users/madmacks/Documents/Arduino/libraries/Talkie/src/Talkie.cpp:561:5: warning: '--' expression of 'volatile'-qualified type is deprecated [-Wvolatile]
  561 |     free--;
      |     ^~~~
/Users/madmacks/Documents/Arduino/libraries/Talkie/src/Talkie.cpp: In member function 'const uint8_t* Talkie::FIFOPopFront()':
/Users/madmacks/Documents/Arduino/libraries/Talkie/src/Talkie.cpp:577:9: warning: '++' expression of 'volatile'-qualified type is deprecated [-Wvolatile]
  577 |         free++;
      |         ^~~~
Multiple libraries were found for "WiFi.h"
  Used: /Users/madmacks/Library/Arduino15/packages/esp32/hardware/esp32/3.0.4/libraries/WiFi
  Not used: /Users/madmacks/Documents/Arduino/libraries/WiFiNINA
Multiple libraries were found for "Network.h"
  Used: /Users/madmacks/Library/Arduino15/packages/esp32/hardware/esp32/3.0.4/libraries/Network
  Not used: /Users/madmacks/Documents/Arduino/libraries/HomeSpan
exit status 1

Compilation error: exit status 1

Changing to ESP32 core 2.0.17 compiles cleanly, but throws the following error at run time.

E (7336) timer_group: timer_isr_callback_add(236): register interrupt service failed
```

### What were you expecting to happen?

I expected the code to "speak" different messages when the water level changes.

### Additional context.

Maybe there's an issue using the Hiletgo ESP32-Wroom-32D and the Espressif core library? I will try with a more generic library and see if that changes things. But shouldn't Talkie work OOTB with the ESP32, as long as I don't use PINs that are used internally based on the ESP32 docs for this library? 

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2024-09-05:
I will check next week

madmacks59, 2024-09-17:
Any feedback yet?

ArminJo, 2024-09-18:
I cannot reproduce it!!!
Do not check checkboxes without thinking.

closed 2024-09-18 by ArminJo

madmacks59, 2024-09-18:
> I cannot reproduce it!!! Do not check checkboxes without thinking.

That’s really helpful. Appreciate the support and conviviality…

I guess all the errors in the compile i listed around timers, and the runtime error I listed, are just figments of my imagination. My obvious/only solution will be to switch to a different sound/talk library. Awesome!