ArminJo/Talkie #50: Set pitch

ErikOostveen, 2024-11-02:
Great job in keeping talkie going!

I would love to see a way of adding the sample rate to the say function - so say(<word>,6000) would play slower and say(<word>,10000) faster with 8000 being the default when not given --- how difficult would this to implement?

ArminJo, 2024-11-03:
Can you first test it?
1. Change #define SAMPLE_RATE 8000 // Speech engine sample rate to the value you want
2. Change #define ISR_RATIO (25000/ (1000000 / SAMPLE_RATE) ) to #define ISR_RATIO 200
3. Test it if it works and the effect is reasonable.

ErikOostveen, 2024-11-03:
Thanks for getting back to quickly!

That test works nicely (I tried 4000 to 16000)

deladriere, 2024-11-03:
Hi Erik also check this one derived from my Talko : it add speed / pitch / and toy bending 😉
https://github.com/technologiescollege/ArduinoTechnoEduc/blob/master/portable/sketchbook/libraries/Talkietz/talkietz.h

ArminJo, 2024-11-03:
@ErikOostveen 
Done 😀
Can you please test the new library version from this repo?
I added parameter `aSampleRateForPitch `.

ErikOostveen, 2024-11-05:
Nice one @ArminJo 

It took me a little while to realize I had to Enable pitch for it to work but after that it seems to work well. Adding these features makes the library instantly more attractive for folks like me looking for ways to manipulate sound :-)

I"ll also have a look at @deladriere

ArminJo, 2024-11-05:
@ErikOostveen 
Plese give me feedback, if the speed and bending features of deladriere's version makes sense for folks like you.
I currently habe not time to test this features for myself.

closed 2024-11-05 by ArminJo

ErikOostveen, 2024-11-05:
> Hi Erik also check this one derived from my Talko : it add speed / pitch / and toy bending 😉 https://github.com/technologiescollege/ArduinoTechnoEduc/blob/master/portable/sketchbook/libraries/Talkietz/talkietz.h and the example https://github.com/technologiescollege/ArduinoTechnoEduc/blob/master/portable/sketchbook/libraries/Talkietz/examples/demo/demo.ino

That was worth looking at - I like being able to also set speed and distortion - with distortion sounding like putting a voice through a rindmodulator - nice work @deladriere !

... so, @ArminJo the library could use adding those two parameters as well - and what about an on-the-fly reverse-speech option?  :-)