ArminJo/Talkie #52: Teensy 4.0

ErikOostveen, 2025-07-09:
There's no output on PIN12 of the Teensy 4.0

Is there a way of making it work for 4.0 ?

PS: I'd like to move from the Nano V3 to the Teensy 4.0 because the Nano only has a tiny 32kb to store program and phonemes/words. The Teensy has got plenty more to even add new libraries.

ErikOostveen, 2025-07-10:
Sorted it (for example set PIN 13 as output):

in the code's setup():
  // Tell Talkie to use pin 13 for its PWM audio:
  Voice.beginPWM(13);

  // Switch to 12-bit output and 62.5 kHz PWM (same as the library normally does):
  analogWriteResolution(13);
  analogWriteFrequency(13, 62500);

in Talkie.cpp:

#  else // Teensy 3.0, Teensy 4.0 / 4.1
#define PWM_OUTPUT_FUNCTION(nextPwm) analogWrite(13, nextPwm)
    analogWriteFrequency(13, 62500);
#  endif // (DAC_PIN)

ArminJo, 2025-07-11:
Is this an issue of the library or a personal one?

closed 2025-07-11 by ArminJo